adafruit/Adafruit_CircuitPython_LSM9DS1 #19: XG requires reboot to function properly

becorey, 2020-01-16:
Hello, I am using the LSM9DS1 connected to an ESP32-PICO-D4 over SPI on a custom board. I found that it would fail when checking `if self._read_u8(_XGTYPE, _LSM9DS1_REGISTER_WHO_AM_I_XG) != _LSM9DS1_XG_ID`, raising the `RuntimeError('Could not find LSM9DS1')`

I fixed it by changing the line above, 
from: `self._write_u8(_XGTYPE, _LSM9DS1_REGISTER_CTRL_REG8, 0x05)` 
to: 
`self._write_u8(_XGTYPE, _LSM9DS1_REGISTER_CTRL_REG8, int('0b10000101', 2))`
This changes the BOOT flag on the register CTRL_REG8 from 0 to 1, telling the XG to reboot memory content. I noticed that in your line to reboot the mag sensor you did set the REBOOT flag, this was the clue to also BOOT the XG.

I am not sure if this is specific to the ESP32-PICO-D4 but wanted to share my solution

caternuson, 2020-01-17:
Can you provide more info on the state of the hardware when you run into this issue. Does it happen every time on first boot? Or only after some other event(s) have occurred?

kattni, 2020-05-04:
Closing this issue due to lack of activity.

closed 2020-05-04 by kattni