adafruit/Adafruit_CircuitPython_LSM9DS1 #24: Looking for burst data from LSM9DS1

memsengineer, 2020-09-23:
I have an application where I would like to get burst data from the LSM9DS1 when it is connected to a Raspberry Pi. As an example I would like to collect 10 seconds of ~900Hz data from the accelerometers and gyros. I think the chip is capable of it. I have not found the capability in the CircuitPython package yet.

Is the capability in the package but I have not found it?
Are there any plans to add the capability?
Is it even possible to do it with Circuit Python?

ladyada, 2020-09-23:
you would have to use the FIFO to do that, and we don't have support for it int he library

memsengineer, 2020-09-23:
Thank you for the quick reply. I will see if I can find another way.

aurelienbl, 2021-01-26:
Hi @memsengineer, 
Following up on this question. I would be interested to know if you have found a solution to probe to sensor at a high frequency.
Best,
Aurélien

memsengineer, 2021-01-26:
What I did was moved to an OS called MBED. I found a library written for the sensor board that I had (LSM9DS1). I had to get a micro-controller board to use with MBED as the raspberry pi was not compatible with MBED. After many hours of fiddling, I got it to work.

aurelienbl, 2021-01-27:
Thanks a lot, I'll have a look.