adafruit/Adafruit_CircuitPython_LSM9DS1 #25: Using SPI gives same measurement for all magnetometer axes

colem404, 2020-10-22:
When I use the code for the SPI, I noticed that all three values were the same when using calling sensor.magnetic. Initially, I believed it to be a very strong magnetic interference, but it behaved as expected after switching to the I2C connection. Haven't yet nailed down what the issue is.

siddacious, 2020-10-26:
This sounds like an issue with the register address auto-increment. I'll take a look.

colem404, 2020-12-19:
Thanks for looking into it. I don't know if you found the issue yet, but I think replacing 0x80 with 0xC0 in line 288 of https://github.com/adafruit/Adafruit_CircuitPython_LSM9DS1/blob/master/adafruit_lsm9ds1.py will fix the issue. I haven't tried it yet and won't get the chance until mind January. But I came to this conclusion from reading the magnetometer section(5.3) of this document (https://www.st.com/resource/en/datasheet/lsm9ds1.pdf). It seems the auto incrementing was the issue just as you said.

siddacious, 2020-12-21:
Hello! sorry for the delay, this must have dropped off my radar. You are right, thats the fix needed. I've pushed a PR with the fix and will close this when it is merged. 

Thank you for your patience.

closed 2020-12-28 by siddacious