adafruit/Adafruit_CircuitPython_LSM9DS1 #26: Sample rate for accelerometer?

maha-haji, 2020-12-02:
Hi,

Is it possible to set the sample rate for the accelerometer? In the Arduino library there is a setting that can be altered, but I don't see the same in this library.

Thanks!

caternuson, 2021-02-15:
Reopening for further discussion. There are some additional features here that do not appear to be currently implemented. Others have requested these also. If there's a reason for not doing this, like making the library too big or something, then let's document it here at least.

lucasgaley, 2021-02-15:
Hello,

From the datasheet, it appears to me that in adafruit_lsm9ds1.py#L152, the gyroscope is set to continuous mode. In that same write, it looks as though the datarate is set to 952 hz (0xC0 or 11000000). If the first three parts of that binary string had an option selection like the acceleration ranges in L#166-191, that would be very beneficial.

Additionally, the magnetometer datarate appears to never be set. From the datasheet, the continuous mode does not set a datarate for the magnetometer, and it's default datarate appears to be 10 hz. The key register would be REG1_M, which is already defined. I can do some work on this (mostly copying the user facing code for the ranges, and hopefully adapting them for the frequencies), but my unfamiliarity would require some oversight to make sure that what I propose makes sense.

lucasgaley, 2021-02-26:
In response to this, I have added all the code that I think would be necessary. I haven't had a chance to test things yet, and I'm not even sure what level of tests would be necessary. Please see below for "my" version.

https://github.com/lucasgaley/Adafruit_CircuitPython_LSM9DS1/blob/master/adafruit_lsm9ds1.py